code-iai/iai_kinect2 #548: Catkin_Make Issue: recipe for target 'all' failed

vec123, 2019-10-27:
Hi, I am a beginner in ROS and trying to build some packages with catkin_make. Somehow I am getting the following error when trying to run catkin_make. 
recipe for target 'all' failed
The first error message shown is:  No rule to make target '/usr/lib/libpython3.7m.so', needed by '/home/victor/catkin_ws/devel/lib/python2.7/dist-packages/tf2_py/_tf2.so'.  Stop
followed by a notification that: recipe for target 'geometry2/tf2_py/CMakeFiles/tf2_py.dir/all' failed.
Does anyone maybe know how to solve this issue? 
Thank you in advance :)

vec123, 2019-10-27:
The full error message is:
[ 52%] Built target tf_generate_messages
make[2]: *** No rule to make target '/usr/lib/libpython3.7m.so', needed by '/home/victor/catkin_ws/devel/lib/python2.7/dist-packages/tf2_py/_tf2.so'.  Stop.
CMakeFiles/Makefile2:3807: recipe for target 'geometry2/tf2_py/CMakeFiles/tf2_py.dir/all' failed
make[1]: *** [geometry2/tf2_py/CMakeFiles/tf2_py.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 58%] Built target tf2_ros
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

fengjiufa, 2019-11-29:
Me too. I also encountered this problem, do not know how to solve。
/usr/bin/ld: warning: libopencv_core.so.3.4, needed by /home/jiufafeng/rosprograms/catkin_ws/devel/lib/libkinect2_registration.so, may conflict with libopencv_core.so.3.2
/usr/bin/ld: warning: libopencv_imgcodecs.so.3.4, needed by /home/jiufafeng/anaconda3/lib/libopencv_datasets.so.3.4.2, may conflict with libopencv_imgcodecs.so.3.2
//usr/lib/x86_64-linux-gnu/libgdcmMSFF.so.2.8：对‘uuid_generate@UUID_1.0’未定义的引用
//home/jiufafeng/anaconda3/lib/libfontconfig.so.1：对‘FT_Done_MM_Var’未定义的引用
//usr/lib/x86_64-linux-gnu/libgdcmMSFF.so.2.8：对‘uuid_parse@UUID_1.0’未定义的引用
//usr/lib/x86_64-linux-gnu/libgdcmMSFF.so.2.8：对‘uuid_unparse@UUID_1.0’未定义的引用
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libSM.so：对‘uuid_unparse_lower@UUID_1.0’未定义的引用
collect2: error: ld returned 1 exit status
iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/build.make:507: recipe for target '/home/jiufafeng/rosprograms/catkin_ws/devel/lib/kinect2_viewer/kinect2_viewer' failed
make[2]: *** [/home/jiufafeng/rosprograms/catkin_ws/devel/lib/kinect2_viewer/kinect2_viewer] Error 1
CMakeFiles/Makefile2:3101: recipe for target 'iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/all' failed
make[1]: *** [iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed