code-iai/iai_kinect2 #549: [DepthRegistrationOpenCL::init] failed to build program: -11

BrooklynBoy21, 2019-10-28:
Hello, I did everything according to the instruction, but when I run:
kinect2_bridge kinect2_bridge.launch 

I am getting next error:

... logging to /home/yehor/.ros/log/e3723f40-f959-11e9-8fbe-f0038c56ac13/roslaunch-egor-13987.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://egor:45047/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/yehor/catki...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_device: -1
 * /kinect2_bridge/reg_method: default
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[kinect2-1]: started with pid [14002]
process[kinect2_bridge-2]: started with pid [14003]
process[kinect2_points_xyzrgb_sd-3]: started with pid [14004]
[ INFO] [1572252162.084529782]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1572252162.088051543]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
process[kinect2_points_xyzrgb_qhd-4]: started with pid [14009]
process[kinect2_points_xyzrgb_hd-5]: started with pid [14019]
[ INFO] [1572252162.102697879]: Initializing nodelet with 4 worker threads.
[ INFO] [1572252162.108853453]: waitForService: Service [/kinect2/load_nodelet] is now available.
[ INFO] [1572252162.190205529]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/yehor/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: default
     depth_device: -1
       reg_method: default
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) HD Graphics Kabylake ULT GT2 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel(R) HD Graphics Kabylake ULT GT2 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
libva info: VA-API version 1.1.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nouveau_drv_video.so
libva info: Found init function __vaDriverInit_1_1
libva info: va_openDriver() returns 0
[Info] [VaapiRgbPacketProcessorImpl] driver: Mesa Gallium driver 19.0.8 for NV118
[Error] [VaapiRgbPacketProcessorImpl] num_entrypoints >= 1 && num_entrypoints <= max_entrypoints failed
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 7 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 085777135147
[Info] [Freenect2Impl] found 1 devices
[ INFO] [1572252163.660786128]: [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [1572252163.660840559]: [Kinect2Bridge::initDevice]   0: 085777135147 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [1572252163.818828866]: [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [1572252164.178993785]: [Kinect2Bridge::initDevice] device serial: 085777135147
[ INFO] [1572252164.179139378]: [Kinect2Bridge::initDevice] device firmware: 4.0.3917.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[ WARN] [1572252164.844688540]: [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [1572252164.844721239]: [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [1572252164.844736003]: [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [1572252164.844767004]: [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [1572252164.929997622]: [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [1572252164.930040244]: [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [1572252164.948490811]: [DepthRegistrationOpenCL::init] devices:
[ INFO] [1572252164.948533506]: [DepthRegistrationOpenCL::init]   0: Intel(R) HD Graphics Kabylake ULT GT2
[ INFO] [1572252164.948551479]: [DepthRegistrationOpenCL::init] selected device: Intel(R) HD Graphics Kabylake ULT GT2
[ERROR] [1572252164.957451171]: [DepthRegistrationOpenCL::init] [depth_registration_opencl.cpp](216) data->program.build(options.c_str()) failed: -11
[ERROR] [1572252164.957520199]: [DepthRegistrationOpenCL::init] failed to build program: -11
[ERROR] [1572252164.957586370]: [DepthRegistrationOpenCL::init] Build Status: -2
[ERROR] [1572252164.957608898]: [DepthRegistrationOpenCL::init] Build Options:	
[ERROR] [1572252164.957691558]: [DepthRegistrationOpenCL::init] Build Log:	 stringInput.cl:190:31: error: call to 'sqrt' is ambiguous
/usr/lib/x86_64-linux-gnu/beignet/include/ocl_math.h:85:20: note: candidate function
/usr/lib/x86_64-linux-gnu/beignet/include/ocl_math.h:184:19: note: candidate function
/usr/lib/x86_64-linux-gnu/beignet/include/ocl_math.h:1431:21: note: candidate function
/usr/lib/x86_64-linux-gnu/beignet/include/ocl_math.h:1432:21: note: candidate function
/usr/lib/x86_64-linux-gnu/beignet/include/ocl_math.h:1433:21: note: candidate function
/usr/lib/x86_64-linux-gnu/beignet/include/ocl_math.h:1434:21: note: candidate function
/usr/lib/x86_64-linux-gnu/beignet/include/ocl_math.h:1435:22: note: candidate function
/usr/lib/x86_64-linux-gnu/beignet/include/ocl_math.h:1436:20: note: candidate function
/usr/lib/x86_64-linux-gnu/beignet/include/ocl_math.h:1437:20: note: candidate function
/usr/lib/x86_64-linux-gnu/beignet/include/ocl_math.h:1438:20: note: candidate function
/usr/lib/x86_64-linux-gnu/beignet/include/ocl_math.h:1439:20: note: candidate function
/usr/lib/x86_64-linux-gnu/beignet/include/ocl_math.h:1440:21: note: candidate function

[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[ERROR] [1572252164.958602251]: [Kinect2Bridge::start] Initialization failed!
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing
[FATAL] [1572252165.061431936]: Failed to load nodelet '/kinect2_bridge` of type `kinect2_bridge/kinect2_bridge_nodelet` to manager `kinect2'
[kinect2_bridge-2] process has died [pid 14003, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/yehor/.ros/log/e3723f40-f959-11e9-8fbe-f0038c56ac13/kinect2_bridge-2.log].
log file: /home/yehor/.ros/log/e3723f40-f959-11e9-8fbe-f0038c56ac13/kinect2_bridge-2*.log
[kinect2_bridge-2] restarting process

./Protonect is working dine and showing 4 screens with all options (gl cl cpu).
Could you please help me? thanks in advance!

PaulSantos8, 2019-10-30:
I have the same problem, the only way it works is using "roslaunch kinect2_bridge kinect2_bridge.launch publish_tf:=true reg_method:=cpu depth_method:=opengl"

because ros uses opencl as a default, but there is some problem with opencl, and the only way it works is using the cpu.

PaulSantos8, 2019-10-30:
Try changing line 190 of kinect2_registration/src/depth_registration.cl
from
const float4 tmp = (float4)(sqrt(2.0));
to
const float4 tmp = (float4)(sqrt(2.0f));

BrooklynBoy21, 2019-10-30:
@PaulSantos8 
Yes, thank you!!!! I also was able to launch it with _reg_method:=cpu.
And your second post is correct, I have changed that and it starts to work, thank you very much.

By the way, do you know what is that problem and why do we need to change it?

max11gen, 2020-02-18:
@PaulSantos8 Did you make a PR for this? Should be fixed in the master, shouldn't it?

Ktysai, 2021-08-01:
> Try changing line 190 of kinect2_registration/src/depth_registration.cl
> from
> const float4 tmp = (float4)(sqrt(2.0));
> to
> const float4 tmp = (float4)(sqrt(2.0f));

great reply, now works with opencl