code-iai/iai_kinect2 #557: would it work at ROS Noetic Ninjemys

MengHao666, 2020-06-17:


ACarfi, 2020-06-19:
It works for me. I had to change few things in the source code to make it work with opencv 4.2 but just minor modifications.

MengHao666, 2020-06-20:
> It works for me. I had to change few things in the source code to make it work with opencv 4.2 but just minor modifications.
> 
> Anyway I am having problems with the reference frame that I am trying to solve right now.

wonderful thing! Could u publish it when u finally successed?
I want to write python3 code with ROS and the KinectV2 device ,So I wanted this repo could work at  ROS Noetic Ninjemys .
But as I compile without any modification,it just failed to get the data of KinectV2

ACarfi, 2020-07-02:
I have uploaded my edited version here: https://github.com/TheEngineRoom-UniGe/iai_kinect2.git
for me it works right now

MengHao666, 2020-07-02:
Well done!
I will try it in the near future and see if it works for me!
Thanks for it in advance!

Ktysai, 2021-08-02:
> I have uploaded my edited version here: https://github.com/TheEngineRoom-UniGe/iai_kinect2.git
> for me it works!

**@ACarfi** 
An interesting solution. I have not tried exactly your code but something similar. For ROS2 implementation you have some solutions?

ACarfi, 2021-08-04:
@Ktysai no I am sorry. I have jet no experience with ROS2.

Fred-Lyu, 2023-03-29:
> I have uploaded my edited version here: https://github.com/TheEngineRoom-UniGe/iai_kinect2.git for me it works!

Ohooo,thank u for this interesting solution

closed 2023-03-30 by MengHao666