code-iai/iai_kinect2 #558: Updating OpenCV 3 support info

gokhansolak, 2020-07-08:
Hello, apparently at one point OpenCV3 got support. I was able to build it with ROS Melodic / Ubuntu 18.04 and run `kinect2_bridge.launch`. The camera runs, I can read the `camera_info`, `image_color` etc and view it using `image_view`. 

However, the readme still says that it won't work:
[image]

Could you please confirm that it is fully supported and update the readme accordingly?

Thank you

Ktysai, 2020-07-14:
I have ROS Melodic and Ubuntu 18. Still in a struggle to make it work. Can you post details?

gokhansolak, 2020-07-15:
Hi Ktysai,

I did not do anything special, besides following the installation instructions. During [libfreenect2 installation](https://github.com/OpenKinect/libfreenect2#linux), I installed the optional OpenCL and VAAPI libs. My system:

ROS-Melodic (standard setup)
Ubuntu 18.04 (everything up-to-date)
Graphics card: Intel Corporation HD Graphics 620 (rev 02)
OpenCV version: 3.2.0

Note: I only run `kinect2_bridge.launch`, no errors. I use bridge outputs, but I did not use any other features of this package, yet.

Ktysai, 2020-07-17:
Thank you, @gokhansolak 

I have installed all on Intel NUC i5, but OpenCL does not work.
What has helped was to run the bridge with those settings`rosrun kinect2_bridge kinect2_bridge _depth_method:=cpu _reg_method:=cpu`