code-iai/iai_kinect2 #559: Error :catkin_make -DCMAKE_BUILD_TYPE="Release"

ChenxDD, 2020-07-23:
Hi, Can you tell me why the following happens when I input “catkin_make -DCMAKE_BUILD_TYPE="Release"
This is the log

gcslam@ubuntu:~/catkin_ws$ catkin_make -DCMAKE_BUILD_TYPE="Release"
Base path: /home/gcslam/catkin_ws
Source space: /home/gcslam/catkin_ws/src
Build space: /home/gcslam/catkin_ws/build
Devel space: /home/gcslam/catkin_ws/devel
Install space: /home/gcslam/catkin_ws/install

Running command: "make cmake_check_build_system" in "/home/gcslam/catkin_ws/build"
Running command: "make -j2 -l2" in "/home/gcslam/catkin_ws/build"
[ 27%] Built target kinect2_registration
[ 36%] Building CXX object iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o
[ 45%] Building CXX object iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/src/kinect2_bridge.cpp.o
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘void Kinect2Bridge::initCompression(int32_t, int32_t, bool)’:
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:464:28: error: ‘CV_IMWRITE_JPEG_QUALITY’ was not declared in this scope
464 | compressionParams[0] = CV_IMWRITE_JPEG_QUALITY;
| ^~~~~~~~~~~~~~~~~~~~~~~
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘void Kinect2Bridge::initCompression(int32_t, int32_t, bool)’:
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:464:28: error: ‘CV_IMWRITE_JPEG_QUALITY’ was not declared in this scope
464 | compressionParams[0] = CV_IMWRITE_JPEG_QUALITY;
| ^~~~~~~~~~~~~~~~~~~~~~~
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:466:28: error: ‘CV_IMWRITE_PNG_COMPRESSION’ was not declared in this scope
466 | compressionParams[2] = CV_IMWRITE_PNG_COMPRESSION;
| ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:466:28: error: ‘CV_IMWRITE_PNG_COMPRESSION’ was not declared in this scope
466 | compressionParams[2] = CV_IMWRITE_PNG_COMPRESSION;
| ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:468:28: error: ‘CV_IMWRITE_PNG_STRATEGY’ was not declared in this scope
468 | compressionParams[4] = CV_IMWRITE_PNG_STRATEGY;
| ^~~~~~~~~~~~~~~~~~~~~~~
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:468:28: error: ‘CV_IMWRITE_PNG_STRATEGY’ was not declared in this scope
468 | compressionParams[4] = CV_IMWRITE_PNG_STRATEGY;
| ^~~~~~~~~~~~~~~~~~~~~~~
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:469:28: error: ‘CV_IMWRITE_PNG_STRATEGY_RLE’ was not declared in this scope
469 | compressionParams[5] = CV_IMWRITE_PNG_STRATEGY_RLE;
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:469:28: error: ‘CV_IMWRITE_PNG_STRATEGY_RLE’ was not declared in this scope
469 | compressionParams[5] = CV_IMWRITE_PNG_STRATEGY_RLE;
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘void Kinect2Bridge::receiveColor()’:
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1103:45: error: ‘CV_BGRA2BGR’ was not declared in this scope
1103 | cv::cvtColor(tmp, images[COLOR_HD], CV_BGRA2BGR);
| ^~~~~~~~~~~
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘void Kinect2Bridge::receiveColor()’:
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1103:45: error: ‘CV_BGRA2BGR’ was not declared in this scope
1103 | cv::cvtColor(tmp, images[COLOR_HD], CV_BGRA2BGR);
| ^~~~~~~~~~~
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1107:45: error: ‘CV_RGBA2BGR’ was not declared in this scope
1107 | cv::cvtColor(tmp, images[COLOR_HD], CV_RGBA2BGR);
| ^~~~~~~~~~~
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1107:45: error: ‘CV_RGBA2BGR’ was not declared in this scope
1107 | cv::cvtColor(tmp, images[COLOR_HD], CV_RGBA2BGR);
| ^~~~~~~~~~~
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘void Kinect2Bridge::processIrDepth(const cv::Mat&, std::vectorcv::Mat&, const std::vectorKinect2Bridge::Status&)’:
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1183:50: error: ‘CV_BGRA2BGR’ was not declared in this scope
1183 | cv::cvtColor(tmp, images[COLOR_SD_RECT], CV_BGRA2BGR);
| ^~~~~~~~~~~
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘void Kinect2Bridge::processIrDepth(const cv::Mat&, std::vectorcv::Mat&, const std::vectorKinect2Bridge::Status&)’:
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1183:50: error: ‘CV_BGRA2BGR’ was not declared in this scope
1183 | cv::cvtColor(tmp, images[COLOR_SD_RECT], CV_BGRA2BGR);
| ^~~~~~~~~~~
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1187:50: error: ‘CV_RGBA2BGR’ was not declared in this scope
1187 | cv::cvtColor(tmp, images[COLOR_SD_RECT], CV_RGBA2BGR);
| ^~~~~~~~~~~
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1187:50: error: ‘CV_RGBA2BGR’ was not declared in this scope
1187 | cv::cvtColor(tmp, images[COLOR_SD_RECT], CV_RGBA2BGR);
| ^~~~~~~~~~~
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘void Kinect2Bridge::processColor(std::vectorcv::Mat&, const std::vectorKinect2Bridge::Status&)’:
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1250:55: error: ‘CV_BGR2GRAY’ was not declared in this scope
1250 | cv::cvtColor(images[COLOR_HD], images[MONO_HD], CV_BGR2GRAY);
| ^~~~~~~~~~~
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘void Kinect2Bridge::processColor(std::vectorcv::Mat&, const std::vectorKinect2Bridge::Status&)’:
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1250:55: error: ‘CV_BGR2GRAY’ was not declared in this scope
1250 | cv::cvtColor(images[COLOR_HD], images[MONO_HD], CV_BGR2GRAY);
| ^~~~~~~~~~~
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1254:65: error: ‘CV_BGR2GRAY’ was not declared in this scope
1254 | cv::cvtColor(images[COLOR_HD_RECT], images[MONO_HD_RECT], CV_BGR2GRAY);
| ^~~~~~~~~~~
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1254:65: error: ‘CV_BGR2GRAY’ was not declared in this scope
1254 | cv::cvtColor(images[COLOR_HD_RECT], images[MONO_HD_RECT], CV_BGR2GRAY);
| ^~~~~~~~~~~
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1258:57: error: ‘CV_BGR2GRAY’ was not declared in this scope
1258 | cv::cvtColor(images[COLOR_QHD], images[MONO_QHD], CV_BGR2GRAY);
| ^~~~~~~~~~~
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1258:57: error: ‘CV_BGR2GRAY’ was not declared in this scope
1258 | cv::cvtColor(images[COLOR_QHD], images[MONO_QHD], CV_BGR2GRAY);
| ^~~~~~~~~~~
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1262:67: error: ‘CV_BGR2GRAY’ was not declared in this scope
1262 | cv::cvtColor(images[COLOR_QHD_RECT], images[MONO_QHD_RECT], CV_BGR2GRAY);
| ^~~~~~~~~~~
/home/gcslam/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1262:67: error: ‘CV_BGR2GRAY’ was not declared in this scope
1262 | cv::cvtColor(images[COLOR_QHD_RECT], images[MONO_QHD_RECT], CV_BGR2GRAY);
| ^~~~~~~~~~~
make[2]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/build.make:63：iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:1042：iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/all] 错误 2
make[1]: *** 正在等待未完成的任务....
make[2]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/build.make:63：iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/src/kinect2_bridge.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:1069：iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/all] 错误 2
make: *** [Makefile:141：all] 错误 2
Invoking "make -j2 -l2" failed

label01, 2020-09-09:
你解决这个问题了吗

paul-shuvo, 2020-10-28:
Getting the same error!

paul-shuvo, 2020-10-28:
Which version of OpenCV are you using?

PARPedraza, 2020-11-18:
Getting the sme error:

/home/user/catkin_ws/src/SDK-Installation-ROS-Wrapper-for-Kinect-v2-on-Ubuntu-18.04/kinect2_calibration/src/kinect2_calibration.cpp: In constructor ‘Recorder::Recorder(const string&, const string&, const string&, const string&, Source, bool, bool, const Size&, float)’:
/home/user/catkin_ws/src/SDK-Installation-ROS-Wrapper-for-Kinect-v2-on-Ubuntu-18.04/kinect2_calibration/src/kinect2_calibration.cpp:113:22: error: ‘CV_IMWRITE_PNG_COMPRESSION’ was not declared in this scope
     params.push_back(CV_IMWRITE_PNG_COMPRESSION);
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/user/catkin_ws/src/SDK-Installation-ROS-Wrapper-for-Kinect-v2-on-Ubuntu-18.04/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘void Kinect2Bridge::initCompression(int32_t, int32_t, bool)’:
/home/user/catkin_ws/src/SDK-Installation-ROS-Wrapper-for-Kinect-v2-on-Ubuntu-18.04/kinect2_bridge/src/kinect2_bridge.cpp:464:28: error: ‘CV_IMWRITE_JPEG_QUALITY’ was not declared in this scope
     compressionParams[0] = CV_IMWRITE_JPEG_QUALITY;
                            ^~~~~~~~~~~~~~~~~~~~~~~
/home/user/catkin_ws/src/SDK-Installation-ROS-Wrapper-for-Kinect-v2-on-Ubuntu-18.04/kinect2_bridge/src/kinect2_bridge.cpp:466:28: error: ‘CV_IMWRITE_PNG_COMPRESSION’ was not declared in this scope
     compressionParams[2] = CV_IMWRITE_PNG_COMPRESSION;
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/user/catkin_ws/src/SDK-Installation-ROS-Wrapper-for-Kinect-v2-on-Ubuntu-18.04/kinect2_bridge/src/kinect2_bridge.cpp:468:28: error: ‘CV_IMWRITE_PNG_STRATEGY’ was not declared in this scope
     compressionParams[4] = CV_IMWRITE_PNG_STRATEGY;
                            ^~~~~~~~~~~~~~~~~~~~~~~
/home/user/catkin_ws/src/SDK-Installation-ROS-Wrapper-for-Kinect-v2-on-Ubuntu-18.04/kinect2_bridge/src/kinect2_bridge.cpp:469:28: error: ‘CV_IMWRITE_PNG_STRATEGY_RLE’ was not declared in this scope
     compressionParams[5] = CV_IMWRITE_PNG_STRATEGY_RLE;
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/user/catkin_ws/src/SDK-Installation-ROS-Wrapper-for-Kinect-v2-on-Ubuntu-18.04/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘void Kinect2Bridge::initCompression(int32_t, int32_t, bool)’:
/home/user/catkin_ws/src/SDK-Installation-ROS-Wrapper-for-Kinect-v2-on-Ubuntu-18.04/kinect2_bridge/src/kinect2_bridge.cpp:464:28: error: ‘CV_IMWRITE_JPEG_QUALITY’ was not declared in this scope
     compressionParams[0] = CV_IMWRITE_JPEG_QUALITY;
                            ^~~~~~~~~~~~~~~~~~~~~~~
/home/user/catkin_ws/src/SDK-Installation-ROS-Wrapper-for-Kinect-v2-on-Ubuntu-18.04/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘void Kinect2Bridge::receiveColor()’:
/home/user/catkin_ws/src/SDK-Installation-ROS-Wrapper-for-Kinect-v2-on-Ubuntu-18.04/kinect2_bridge/src/kinect2_bridge.cpp:1103:45: error: ‘CV_BGRA2BGR’ was not declared in this scope
         cv::cvtColor(tmp, images[COLOR_HD], CV_BGRA2BGR);
                                             ^~~~~~~~~~~
/home/user/catkin_ws/src/SDK-Installation-ROS-Wrapper-for-Kinect-v2-on-Ubuntu-18.04/kinect2_bridge/src/kinect2_bridge.cpp:1107:45: error: ‘CV_RGBA2BGR’ was not declared in this scope
         cv::cvtColor(tmp, images[COLOR_HD], CV_RGBA2BGR);
                                             ^~~~~~~~~~~
/home/user/catkin_ws/src/SDK-Installation-ROS-Wrapper-for-Kinect-v2-on-Ubuntu-18.04/kinect2_bridge/src/kinect2_bridge.cpp:466:28: error: ‘CV_IMWRITE_PNG_COMPRESSION’ was not declared in this scope
     compressionParams[2] = CV_IMWRITE_PNG_COMPRESSION;
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/user/catkin_ws/src/SDK-Installation-ROS-Wrapper-for-Kinect-v2-on-Ubuntu-18.04/kinect2_bridge/src/kinect2_bridge.cpp:1107:45: note: suggested alternative: ‘CV_RGB’
         cv::cvtColor(tmp, images[COLOR_HD], CV_RGBA2BGR);
                                             ^~~~~~~~~~~
                                             CV_RGB
/home/user/catkin_ws/src/SDK-Installation-ROS-Wrapper-for-Kinect-v2-on-Ubuntu-18.04/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘void Kinect2Bridge::processIrDepth(const cv::Mat&, std::vector<cv::Mat>&, const std::vector<Kinect2Bridge::Status>&)’:
/home/user/catkin_ws/src/SDK-Installation-ROS-Wrapper-for-Kinect-v2-on-Ubuntu-18.04/kinect2_bridge/src/kinect2_bridge.cpp:1183:50: error: ‘CV_BGRA2BGR’ was not declared in this scope
         cv::cvtColor(tmp, images[COLOR_SD_RECT], CV_BGRA2BGR);
                                                  ^~~~~~~~~~~
/home/user/catkin_ws/src/SDK-Installation-ROS-Wrapper-for-Kinect-v2-on-Ubuntu-18.04/kinect2_bridge/src/kinect2_bridge.cpp:1187:50: error: ‘CV_RGBA2BGR’ was not declared in this scope
         cv::cvtColor(tmp, images[COLOR_SD_RECT], CV_RGBA2BGR);
                                                  ^~~~~~~~~~~
/home/user/catkin_ws/src/SDK-Installation-ROS-Wrapper-for-Kinect-v2-on-Ubuntu-18.04/kinect2_bridge/src/kinect2_bridge.cpp:1187:50: note: suggested alternative: ‘CV_RGB’
         cv::cvtColor(tmp, images[COLOR_SD_RECT], CV_RGBA2BGR);
                                                  ^~~~~~~~~~~
                                                  CV_RGB
/home/user/catkin_ws/src/SDK-Installation-ROS-Wrapper-for-Kinect-v2-on-Ubuntu-18.04/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘void Kinect2Bridge::processColor(std::vector<cv::Mat>&, const std::vector<Kinect2Bridge::Status>&)’:
[... cut ...]
make[2]: *** [SDK-Installation-ROS-Wrapper-for-Kinect-v2-on-Ubuntu-18.04/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/src/kinect2_bridge.cpp.o] Error 1
CMakeFiles/Makefile2:2622: recipe for target 'SDK-Installation-ROS-Wrapper-for-Kinect-v2-on-Ubuntu-18.04/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/all' failed
make[1]: *** [SDK-Installation-ROS-Wrapper-for-Kinect-v2-on-Ubuntu-18.04/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

I use:
Ubuntu 18.04
PCL 1.8
OpenCV 4.0.0
ROS Melodic

Any idea??

PARPedraza, 2020-11-18:
> Which version of OpenCV are you using?

which is the ideal version?

paul-shuvo, 2020-11-18:
The issue is you are using opencv4. I also came across same issues. You may want to look at this [repo ](https://github.com/paul-shuvo/iai_kinect2_opencv4) which is a fork of this repo and supported for opencv4 (disclaimer: I'm maintaining the repo)

PARPedraza, 2020-11-18:
> The issue is you are using opencv4. I also came across the same issues. You may want to look at [this ](https://github.com/paul-shuvo/iai_kinect2_opencv4) which is a fork of this repo and supported for opencv4 (disclaimer: I'm maintaining the repo)

The problem is resolved, thank you.
But have a new error:

[ERROR] [1605738694.750851635]: [DepthRegistrationOpenCL::init] could not find any suitable device

Any idea?

PARPedraza, 2020-11-18:
> > [ERROR] [1605738694.750851635]: [DepthRegistrationOpenCL::init] could not find any suitable device
> 
> Have you checked this [issue](https://github.com/code-iai/iai_kinect2/issues/371#issue-210152403)?

I read this post but found a solution. :(

paul-shuvo, 2020-11-18:
> > > [ERROR] [1605738694.750851635]: [DepthRegistrationOpenCL::init] could not find any suitable device
> > 
> > 
> > Have you checked this [issue](https://github.com/code-iai/iai_kinect2/issues/371#issue-210152403)?
> 
> I read this post but didn't find a solution. :(

Did you install OpenCL and are you able to run libfreenect2 using `./bin/Protonect` (libfreenect2 installation folder).

PARPedraza, 2020-11-19:
> Did you install OpenCL and are you able to run libfreenect2 using `./bin/Protonect` (libfreenect2 installation folder).

Yes, this is the problem. Cannot run OpenCL: ./bin/Protonect cl. And install driver's [issue](https://www.amd.com/en/support/apu/amd-series-processors/amd-a10-series-apu-for-laptops/a10-7300-radeon-r6-graphics)

I have AMD10 Raedom R6 Graphics. I follow the tutorial [issue](https://drivers.amd.com/relnotes/amd-catalyst-graphics-driver-installer-notes-for-linux-operating-systems.pdf), and I get the error:

error: Detected X Server version 'XServer 1.19.6_64a' is not supported. Supported versions are X.Org 6.9 or later, up to XServer 1.10 (default:v2:x86_64:lib32:XServer 1.19.6_64a:none:4.15.0-106-generic:)
Installation will not proceed.

On the [forum](https://forums.centos.org/viewtopic.php?f=48&t=65366) use Radeon™ Software for Linux® version 18.20 for Ubuntu 18.04; they install:

sudo dpkg -i fglrx-core_14.500-0ubuntu1_amd64_UB_12.04.4.deb
fglrx_14.500-0ubuntu1_amd64_UB_12.04.4.deb 
fglrx-dev_14.500-0ubuntu1_amd64_UB_12.04.4.deb 
fglrx-amdcccle_14.500-0ubuntu1_amd64_UB_12.04.4.deb

you any idea what the correct files?

PARPedraza, 2020-12-08:
> Did you install OpenCL and are you able to run libfreenect2 using `./bin/Protonect` (libfreenect2 installation folder).

Paul,

The iai_kinect2_opencv4, your can install on Ubuntu 20.04 (aka Focal) == ROS Noetic?

I hope, you can help me.

paul-shuvo, 2020-12-08:
> Paul,
>
> The iai_kinect2_opencv4, your can install on Ubuntu 20.04 (aka Focal) == ROS Noetic?
>
> I hope, you can help me.

Yes, I'm currently using it on ROS noetic.